iip-ecosphere/platform #10: Integrate Container generation into plattform/app instantiation

eichelbe, 2022-04-06:
(Integrate) scripts for 
 - Dockerfile
 - Python requirements
 - installed dependencies
 - creation of containers

eichelbe, 2022-10-20:
Containers now go for appName/deviceContainerName/appVersion to separate between apps/devices. For now, ECSdevice enables container creation and has the container name "dflt". PLCnext if present would have "plcNext".

Ports for containers are a bit tricky:
 - By (current) default, ECSruntime and serviceManager may grab ay free port for AAS implementation server (in IVML aasImplServer.devicesAsEphemeral). While this is convenient for testing, it is a problem in a container because the port is determined at runtime and not exposed. For this purpose, the container descriptor has now one or two ports (ContainerType::Ecs_Svc_App, ecs and svcmgr as different processes in the same container) that are assigned by the service manager on startup. Pre-installed containers such as for ECSruntime have to care for determining iip.port and exposing it as part of the container installation on the device.
 - If aasImplServer.devicesAsEphemeral is false, aasImplPort in EcsDevice is consulted. If specified, that port will be taken, if not, it is by default aasImplServer.port. If ContainerType::Ecs_Svc_App, we need two ports. Here, the instantiation currently just takes the specified port (aasImplPort or aasImplServer.port) and adds 1 for the serviceManager port (iip.port.svgMgr).

AhmadAlamoush, 2023-02-21:
The list is completed.

closed 2023-02-21 by AhmadAlamoush