iip-ecosphere/platform #9: Test EASy Docker commands

eichelbe, 2022-04-06:
Already ongoing.

eichelbe, 2022-04-06:
Monika:
- dockerBuildImage working
- dockerImageName working
- dockerSaveImage fails as Java process
- dockerLoadImage not resolvable (in CI with 966d69baade438c3d6b15881e4e4691aa80265bc)

eichelbe, 2022-04-06:
Monika:
- dockerBuildImage working
- dockerImageName working
- dockerSaveImage fails as Java process
- dockerLoadImage working

eichelbe, 2022-06-06:
May not be needed if we go for a local docker repository -> layer reuse and transport

eichelbe, 2022-07-13:
Monika: dockerSaveImage working now, result is tar

eichelbe, 2022-07-14:
@dzikaswinia Monika: dockerPush working in external example with public/private registry. Taking over code. Final validation still to be done, also with auth, but we advanced one step here :)

eichelbe, 2022-08-02:
dockerLogin(String registry, String user, String password)
dockerPushImage(String imageName, String registry, String tag)
dockerLogout(String registry)
dockerLogout()

dzikaswinia, 2022-08-18:
- all commands validated
- dockerPush is working with auth