iip-ecosphere/platform #40: Simplify build process for examples

eichelbe, 2022-06-09:
already on the way :) 
- only one POM
- build.sh
- README.md
- new version of platform handbook

eichelbe, 2022-07-01:
examples.vdw seems to be different/cause problems @AhmadAlamoush

eichelbe, 2022-07-03:
Also Install (change handbook install and README.md accordingly), consider resources folder, synch with #22 to enable new authentication for broker @AhmadAlamoush

eichelbe, 2022-07-03:
Further, can we have a Maven system property to define the name of the resources folder ("resources" vs. "resources.ipr") to switch on demand with "resources" as default, e.g., -Diip.resources=resources.ipr on the Maven command line @AhmadAlamoush

eichelbe, 2022-07-12:
JFYI, upgraded hm22 build process to maven profiles

eichelbe, 2022-07-31:
Question: Shall we stay with "Example"/"EasyGen" for the profile names or - thinking about the service workshop - uniformly switch to "App"/"EasyGen"...

eichelbe, 2022-07-31:
In some cases an explicit default profile may be helpful (must not be in all cases): https://stackoverflow.com/questions/11824328/default-build-profile-for-maven

AhmadAlamoush, 2022-08-01:
> examples.vdw seems to be different/cause problems @AhmadAlamoush Jenkins does not run the code as regression test, because it is not allowed to contact VDW Umati (in discussion with Uwe/RZ). Thus, difficult to keep eye on. Bug #60 fixed.

examples.vdw is completed

> impl.impl/impl.model? @AhmadAlamoush 

Completed

AhmadAlamoush, 2022-08-05:
> Question: Shall we stay in the POM files with "Example"/"EasyGen" for the profile names or - thinking about the service workshop - uniformly switch to "App"/"EasyGen"...

Completed

AhmadAlamoush, 2022-08-05:
> Further, can we have a Maven system property to define the name of the resources folder ("resources" vs. "resources.ipr") to switch on demand with "resources" as default, e.g., -Diip.resources=resources.ipr on the Maven command line @AhmadAlamoush

Completed, Update the pom.xml file for examples.python, examples.KODEX, examples.rtsa, and examples.vdw to have a default value for iip.resources (which is resources) and allow to pass new value by adding (-Diip.resources="NewFolderName") to the command.

eichelbe, 2022-08-08:
Cool. Did you mention that already in the README.md? Can you write a short paragraph for the platform handbook explaining the unified build commands and their options (such as -Diip.resources) - as a summary of Section 6.4 where you already updated the figure?

eichelbe, 2022-08-10:
As discussed yesterday

closed 2022-08-10 by eichelbe