iip-ecosphere/platform #50: Check app start via spring service manager

eichelbe, 2022-07-01:
Monika: Runs in strange spring state

eichelbe, 2022-07-06:
We just had the same behavior in tests when due to an exception in VAB/BaSyx at service shutdown the service instance/JVM was left over. For the tests, we allow to clean the service manager and try to get rid of left-over instances.

@dzikaswinia Did this happen on your side right from the beginning or did you run an application before. You stopped it probably with all... may be we shall take specific care in this case to really get rid of all instances. Would be a duty of the service manager. What do you think?

eichelbe, 2022-07-12:
@dzikaswinia Checked it with actual version... works for me. On the way, some fixes for resource loading #24, resource inclusion of identity store #22 and a superfluous exception-throwing in the Python integration were fixed. Not all are through CI - will require 2-3 commits. Might be, you did something specific, a re-start without prior remove or so?

dzikaswinia, 2022-07-12:
I've never run the example.python app because I was never able to build it.

eichelbe, 2022-07-13:
The service manager accidentally permitted unneeded services instances to become active (spring function definition). Turned to be more local (last commit).

eichelbe, 2022-07-27:
Seems to be more likely on Linux than on Windows

eichelbe, 2022-08-02:
Original message:

ERROR: App with deploymentId [SimpleMeshContainerApp.SimpleTransformer] with state [deployed] doesn't match expected state [unknown]

eichelbe, 2022-08-15:
Seems to be gone (with additional cleanup code, service transitions). Closing.

closed 2022-08-15 by eichelbe