iip-ecosphere/platform #54: Document example models

eichelbe, 2022-07-02:


eichelbe, 2022-07-02:
Could you please add parts of the prepared documentation (pptx and pdf) to examples.KODEX, examples.rtsa and examples.python. For the graphical style, please see configuration.configuration/src/test/easy/summary.pptx. Please link then the respective PDF from the README.md of the individual example projects.

eichelbe, 2022-09-19:
Thanks... nice.

Could you please turn the links in README.md into real links and move the slides into an own docs folder... the resources folder is for resources to be included by the automated integration into the services artifact. And, could you please add similar foles to examples.vdw (now a real application graph rather than just a connector). 

May be afterwards @AhmadAlamoush can add some of his explaining slides on the models etc.

Chris-Sauer, 2022-09-21:
README.ms links fixed, slides moved to new "docs" folder for each example. Working on the examples.vdw silde(s) now.

eichelbe, 2022-10-13:
As discussed, KODEX needs update due to DB upgrade. You may close the task, you may also close it after the KODEX doc update.

Chris-Sauer, 2022-10-18:
Task closed, Documentation for all examples integrated.

eichelbe, 2022-11-21:
The console and broker issue. Please have a look at examples.pythonSync

Chris-Sauer, 2022-11-21:
Updated examplesPythonSync slides and the readme

eichelbe, 2022-11-21:
Why updating examples.pythonSync - that was ok (based on the discussion of last weak), and now has examplePython and examplePythonSync in the doc folder?
Please check/update
- [ ] examples.python
- [ ] examples.vdw
- [ ] example.rtsa
- [ ] and now also examples.pythonSync

Chris-Sauer, 2022-12-05:
Updated the slides to all for examples

closed 2022-12-05 by Chris-Sauer