iip-ecosphere/platform #59: Setup distributed testing environment

eichelbe, 2022-07-05:


eichelbe, 2022-07-05:
* Aquire VMs (-> Uwe, Constantin)
* Specify involved resource VMs (config file)
* Instantiate platform (based on config file, replace IPs)
* Distribute platform to resource VMs (SSH, pre-shared key)
* Build applications (selected from config file)
* Copy deployment plans -> instantiated platform, resource VMs load on server/AAS request applications
* Start applications via CLI/UI based on given deployment plans

eichelbe, 2022-07-05:
Local Docker repo in open source/community license?

eichelbe, 2022-07-07:
Indeed, registry is fully sufficient. A bit more for the installation manual.... besides docker, get the container and install it ;)

JimmyAppel, 2022-07-07:
I successfully installed the open source docker components under windows and setup a local registry which works as inteden. The stepts to get to that point are documented and I will now look into setting up the remote access to the repository.

eichelbe, 2022-07-07:
Great :) @dzikaswinia for your information.... plan is to install it once on our premises and make it available to you, also for UI testing @ChrisKHil

JimmyAppel, 2022-07-08:
Update: I am unable to use a local IP Adress i.e. 192.168.5.5:5002 or other similar ones as docker refuses to listen to it. It ends with: "bind: cannot assign requested address". Something along those lines seemd to be a problem but should have been fixed as far as I can tell.

eichelbe, 2022-07-08:
Anything else running on 5002? Would another port also work (telling docker it shell use another port)?

JimmyAppel, 2022-07-08:
NetStat showed nothing on 192.168.5.5 or 5002. It only workes on localhost for me an at this point im unsure where the error lies. Another way could be to use a domain and make it more or less remote accessable but I do not have a useable domain and thinks that this would not be desireable in the field.

eichelbe, 2022-07-08:
Might be native Linux on the target VM works better? ;)

eichelbe, 2022-07-12:
JFYI: VTL now has also docker push for given registry address as well as login/logout for authentication.

eichelbe, 2022-08-15:
Not realistic for TddT'22, testing on real hardware. Shifted to next milestone.

eichelbe, 2022-09-09:
Platform github now contains folder for test/evaluation environment (_tests/test.environment_). Please start commiting source code/scripts there.

eichelbe, 2022-10-27:
Commit of sources is a project in tests/test.environment. Please move sources up by one level so that project is directly in tests/test.environment, not tests/test.environment/IIPPlatformTesting

closed 2023-04-04 by eichelbe