iip-ecosphere/platform #62: Check AAS cleanup after ECS-Runtime/ServiceManager termination

eichelbe, 2022-07-10:
Shutdown hooks may not work properly/take too long. "Heartbeat" based on monitoring channel might be observed in "platform" component to check for disappearing instances. AAS deletion may be a bit tricky when component comes back, but then an additional check may prevent component startup to be on the safe side.

eichelbe, 2022-07-18:
prepared but not yet activated. Needs testing, also blocking of upcoming AAS while cleanup.
@AhmadAlamoush How did you terminate the ECS-Runtime/Service Manager? kill -SIGINT (aka CTRL-C), kill, kill -9 (there would probably not be any cleanup time for the processes)?

eichelbe, 2022-08-02:
After discussion: It was a kill -9. We will observe the behavior with kill -SIGINT

eichelbe, 2022-08-15:
Needs testing and observation (#59). As long as not urgent, shifted into next milestone.

AhmadAlamoush, 2022-08-24:
Solved. 
To kill the process gracefully and cleanup the processes:
* Linux: By using "kill -SIGINT pid" in command line, while using "kill pid" in the scripts.

* Windows:  By using  [SendSignalCtrlC.zip](https://github.com/iip-ecosphere/platform/files/9415622/SendSignalCtrlC.zip) file to kill the process gracefully with the command "SendSignalCtrlC64.exe pid".

P.S. Reference for SendSignal Project (http://web.archive.org/web/20170909040729/http://www.latenighthacking.com/projects/2003/sendSignal/), a modified and compiled version of SendSignal to send Ctrl-C by (https://stackoverflow.com/questions/813086/can-i-send-a-ctrl-c-sigint-to-an-application-on-windows/29278113#29278113).

eichelbe, 2022-08-24:
Sorry... the heartbeat-based cleanup in the platform is so far not tested. Deferred to next milestone.

After TDDT, plase add unpacked sendSignal to new tools folder of Install package and change build-jk.xml to include the folder into the zip files (usually not required for tar.gz files)

closed 2023-05-06 by eichelbe