iip-ecosphere/platform #63: Turn JSON monitoring structures in AAS to Double with semanticId.

eichelbe, 2022-07-13:
initial solution, intended to contain the unit and the description. Either we turn this into AAS structure or a simple double value with associated semanticId (multi-language description may be added). 

- Prepared in platform. Currently, switched off, i.e., presented as usual.
- Update of Mgt UI required. Would be good to read out structure if type is string, value only if type is Double. If there is a semanticId IRDI, take over unit/description from IRDI/ECLASS.

eichelbe, 2022-07-21:
as basis: semanticId resolution from #68 and AAS operation defined there

eichelbe, 2022-08-09:
As suggested by @ChrisKHil switched in platform code to numeric values with semanticIds.

eichelbe, 2022-08-19:
At least values are still there. No units/semantic ids.

eichelbe, 2022-08-21:
BTW, please adjust the colors for device display to increase readability.

eichelbe, 2023-01-27:
Added Monika, we discussed the semanticId resolution some weeks ago...

dzikaswinia, 2023-02-28:
Mgt UI updated - displaying value, unit and semantic Id.

closed 2023-02-28 by dzikaswinia