iip-ecosphere/platform #66: Check examples.KODEX for fluctuating "_"

eichelbe, 2022-07-15:


eichelbe, 2022-07-15:
Strange Vodoo... first no "_", with -U generate-sources there it is. No compile error. Maven update, ups there it is. Trying to "convince" Jenkins. Then it's your turn.

eichelbe, 2022-07-15:
a bit console mvn update on Jenkins, now running through with "_". Strange, very strange.

eichelbe, 2022-07-15:
some warn that code with "_" may be broken as it initiates Java Vodoo, others just say it's a matter of style and consistency, but not of function. Very strange. Even VIL toIdentifier allows $kip and _kip to pass without modification. Trying to sharpen the identifier translation in the code generation just that whitespaces come around... then, problably, close the issue.

eichelbe, 2022-07-15:
identifiers change ran through, examples still running... closing this issue

closed 2022-07-15 by eichelbe

Chris-Sauer, 2022-07-15:
Okay, will check locally

eichelbe, 2022-07-18:
Strange thing... after Eclipse restart the compile problem was back. After full Maven build with -U, Eclipse close/reopen project it was gone again.

Chris-Sauer, 2022-07-18:
The latest changes +Maven building with -U solved the issue for my locale instance

eichelbe, 2022-07-20:
Happened again after a Maven update of another project :/ Same game, ran generation, closed Eclipse project, re-opened it, gone. Tried to cleanup outdated Maven snapshots with MvnCentral->CleanMvnSnapshots. Let's see whether the ghost comes back.

Chris-Sauer, 2022-07-20:
Pulled latest changes, Eclipse restart, recompiled with -U, installed with -U, no problems, example runs without problem.

eichelbe, 2022-07-21:
Good news, thanks :)