iip-ecosphere/platform #70: Progress reporting UI

eichelbe, 2022-07-21:
Uptake of #67

eichelbe, 2022-10-10:
@ChrisKHil The AAS containing the deployment plan operations shall now have deployPlanAsync and undeployPlanAsync, both returning a taskId. This taskId shall be passed along with the StatusMessages for this task. The task status can be obtained from the AAS via getTaskStatus(taskId)... I missed resetting the task status. Will be committed next.

ChrisKHil, 2022-11-09:
Progress should be should at the bottom of the UI. Unfortunately I could not test if this works as intended. Verificvation needed

eichelbe, 2022-11-10:
Thanks. We will check this as soon as the new UR-PLC is validated and the application can be completely started on the demonstration devices.

eichelbe, 2023-05-09:
refined status tracking is on the way...

eichelbe, 2023-09-08:
working...

closed 2023-09-08 by eichelbe