iip-ecosphere/platform #71: Test examples with MQTT

eichelbe, 2022-07-26:
See platform handbook for model change
automatic (in regression) or manual

AhmadAlamoush, 2022-12-13:
When starting the platform, it shows the following error :
"Cannot create transport connector: Unable to connect to server"
"No transport setup, will not listen to trace records."

eichelbe, 2022-12-31:
Tried with examples.pythonSync and MQTTv3. Fixed generation of application.yml, runs with generated MQTT broker on Windows, fails on Linux with similar problem - connect timed out.

eichelbe, 2023-01-01:
It could be a different setup of the protocols. The MQTT tests operate with actionTimeout=3000 while AMQP uses the default 1000 ms. So far, this is not reflected in the configuration. May also be that some other setup of broker/protocol, e.g., in the K8S experiments is missing...

eichelbe, 2023-01-01:
Indeed, it was the actionTimeout, at least for examples.pythonSync. Default for MQTT is now at least 3000 ms. examples.pythonSync now running with MQTT v3.

commit 2fa142a8 (2023-02-02): mqtt host arg fix #71
closed 2023-02-02 by eichelbe