iip-ecosphere/platform #87: Python: Example for checking python syntax in Maven build process

eichelbe, 2022-08-28:
- [ ] for service.environment
- [ ] for kiServices.function
- [ ] for hm'22 -> examples

JimmyAppel, 2022-09-09:
Ok I am looking into it, I might need further constraints. Our build currently has 4 mvn steps, generate-sources, noDeps, install, genApps. At what stage shall the syntax check happen?
Further is the solution required to be OS agnostic? The one idea I got would probably break under linux as I would be executing "python -m py_compile program_name.py" through mvn exec, but it does not look through Environment variables under linux ( at least according to documentation). Also an unsolved issue would be that python not always is called python...

eichelbe, 2022-09-09:
;)

1) generate-sources downloads/updates the IVML meta model: no
2) genAppsNoDeps generates the interfaces Java/Python and deploys them: no
3) install compiles services and installs them: yes
4) genApps generates the integration code Java/Python, integrates 2) and 3) and deploys it: no

For a first step, extension of the POM for a local compile e.g. in HM22 would be fine. We can this then take over in template generation #92 

If possible, shall be OS agnostic, assuming that a proper Python is installed. Indeed, your idea is similar to the python compilation steps in configuration.configuration. Also the python naming (with a specific path on Jenkins) is also a problem, which is taken into account by the service.environment. So if there is no existing applicable solution, one could be that we package that as an own Maven plugin based on some IIP-E libraries like service.environment. 

But please feel free to explore a bit...

JimmyAppel, 2022-09-12:
Ok I build a solution which should work everywhere the usual python enviroment works. I use your PythonUtils.getPythonExecutable() to get the concrete version. Its not a very nice solution IMO but I lack a better alternative to dynamically discern the concrete python version.
The test itself is inside the src/test... but not as a unit test as I wanted to run this test despite -DsktipTests under windows. The test gets executed using the codehaus.exec plugin thus the new install command: "mvn -P App install exec:java@py-test -DskipTests" and checks one python file which needs to be definde in the POM. Further it can fail the build if desired, this can also be adjusted in the POM.
It tries to utilise "pyflakes" to check the script as it seems to catch quiet a few things. (It will not fail the build if some imports are unused but will inform about it.)
IF the python does not have pyflakes I will run a backup solution using py_compile which can at least catch a few things but seems overall less reliable.

eichelbe, 2022-09-12:
Ok, -DskipTests makes sense and it's simpler to realize then a full maven plugin. Let's collect some experience with that approach as it must be called explicitly - similar to the generated tests.

eichelbe, 2022-09-22:
Integrated, now globally in services.environment to facilitate reuse. No main-arg documentation so far although used in maven example. No tests so far -> service.environments. Integrated into template generation #92, please consider for service integration workshop.

closed 2022-09-27 by JimmyAppel