iip-ecosphere/platform #110: UI complains

dzikaswinia, 2022-12-01:
While running UI in container following warning occurs:

```
device/mgtUi
`-- (empty)

npm WARN saveError ENOENT: no such file or directory, open '/device/mgtUi/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/device/mgtUi/package.json'
npm WARN mgtUi No description
npm WARN mgtUi No repository field.
npm WARN mgtUi No README data
npm WARN mgtUi No license field.

+ express@4.18.2
added 57 packages from 42 contributors and audited 57 packages in 3.731s

7 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

/device/mgtUi
`-- (empty)

npm WARN saveError ENOENT: no such file or directory, open '/device/mgtUi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/device/mgtUi/package.json'
npm WARN mgtUi No description
npm WARN mgtUi No repository field.
npm WARN mgtUi No README data
npm WARN mgtUi No license field.

+ cors@2.8.5
added 2 packages from 2 contributors and audited 105 packages in 0.885s

7 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

Server listening on port 4200
```

Alpine Linux v3.13
npm v6.14.17
node v14.20.1

dzikaswinia, 2023-02-28:
The warning is still there (using the new version of mgtUI)

eichelbe, 2023-05-09:
I've never seen it so far.... is it still there (after some npm/angular updates)?

closed 2025-10-07 by AhmadAlamoush