iip-ecosphere/platform #111: Simplify platform installation

eichelbe, 2022-12-04:
- [ ] try installing platformDependencies in a single step 
- [ ] use Mvn platform instantiator #109 with properties that can be set optionally from CLI
- [ ] clean up platformDependencies folder
- [ ] check / update github action script
- [ ] update platform handbook
- [ ] update installation slides
- [ ] clean up pom?

eichelbe, 2022-12-05:
@dzikaswinia @AhmadAlamoush The one-command install (`mvn install -Dunpack.force=true`) seems to be working. The old install shall be working, but the POM needs a shift of model-unpacking to generate-sources (via the extended IIP-Ecosphere unpack plugin) and as `src/main/easy` exists also `-Dunpack.force=true`. Explicit installation of `platformDependencies` is then not needed anymore.

I did not commit this so far as it may interfere with your efforts on #59 #102 #91, so I would need an "approval" ;)

eichelbe, 2022-12-05:
@dzikaswinia @AhmadAlamoush after extending the IIP unpack plugin by a list of `initiallyAllowed` files, the `-Dunpack.force=true` is only needed to update the model. 

If you want to try it yourself, the patch for the POM is attached.
[install.zip](https://github.com/iip-ecosphere/platform/files/10155661/install.zip)

AhmadAlamoush, 2023-01-04:
The installation slides are Updated.

closed 2023-01-04 by AhmadAlamoush