iip-ecosphere/platform #135: ECSRuntime container generation - allow Docker to access local docker registry

eichelbe, 2023-03-16:
Container generation, if there is a docker registry configured, add in the ECS-Runtime containers

{
  "insecure-registries" : ["_ip_:_port_"]
}

to /etc/docker/docker for docker-out-of-docker starts of application containers.

As we plan for AllApps containers, this is potentially not needed for HM'23. For now, we keep it on the wish-list.

AhmadAlamoush, 2023-03-30:
It is not possible to move daemon.json inside the docker container (docker-out-of-docker) since there is no daemon inside of the docker container. 
It is linked to the daemon of the Docker installed on the device.

eichelbe, 2023-03-31:
In the DooD-Settings of EcsRuntime, where the EcsRuntime as container manager starts external Docker containers??? There is no Docker??? DooD was not working before without Docker inside the container. Are you sure???

eichelbe, 2023-04-03:
Adding Monika. May be Monika has some additional ideas...

AhmadAlamoush, 2023-04-03:
The way DooD works is by linking the docker.sock using -v /var/run/docker.sock:/var/run/docker.sock, the containers that are created from inside of the container are accessible from outside of the container. All of them are using the same daemon to work with the containers and images. 

I think there is docker functionality (commands) inside of the container but without running daemon, since it is DooD.

eichelbe, 2023-04-03:
Depending on the Java library, it may be the case that even Docker inside the container is not needed anymore. Might be, the information that we got was missing some parts. Anyway, would `--insecure-registry` be an option?

AhmadAlamoush, 2023-04-03:
For `--insecure-registry`, there should be a daemon to add the `--insecure-registry` to it.
So without daemon we can't add it inside of the container.

eichelbe, 2023-04-03:
Looked to me that docker pull would support that. So also that option is gone. TLS?

AhmadAlamoush, 2023-04-03:
I tested the TLS, but the same thing. 
Without a daemon inside of the container, we can't access the local docker registry with or without a security option.

eichelbe, 2023-04-03:
Well, which options do we have if we cannot change the docker daemon?

AhmadAlamoush, 2023-06-12:
It is not possible to change insecure-registries inside of the container. 
We solve it through the install scripts for the platform, in issue #107

closed 2023-06-12 by AhmadAlamoush