iip-ecosphere/platform #146: UI: Editors

eichelbe, 2023-04-28:
Also as a reminder... as discussed with Monika, enable the "edit" buttons of the lists by opening an own editor, for now for
- [ ] constants
- [ ] data types
- [ ] nameplates (see #145)
- [ ] services
- [ ] servers
- [ ] applications

List may not be complete (basic settings not included for now). Meshes have own editor. The editor (dialogs) for now may go without calling the respective AAS operation, i.e., changing the underlying model.

ChrisKHil, 2023-04-29:
What input fields should be provided  for each Editor?
Im thinking of implementing the Editors as a modal view (Editor in a Box that is overlayed on top of the list Page where the Editor is opened).

eichelbe, 2023-05-02:
Basically, all input fields per type in the configuration submodel. As instances occur there only when they do exist (empty model looks pretty different) I'm thinking about providing the "meta" information once per type in that sub-model (we shall discuss where and how). This may allow you for automatically deriving a dialog (probably looks pretty ugly) or at least to indicate the declared but missing fields. The types for the individual fields (already there, or then via "meta" types) would allow you for creating a factory of editor elements so that you can reuse the creation process. Also all elements in the submodel shall have a description that can be used as help text for the editors.

Let's discuss that today in our meeting.

eichelbe, 2023-05-02:
As discussed and decided, we add a "meta" SMEC to the configuration AAS representing the complex (IVML compound) types. As far as possible, we try to go with automatically derived editors.

eichelbe, 2023-05-02:
ok, "meta" is now in the configuration submodel. Let's see whether this is ok, sufficient, ...

closed 2023-09-30 by eichelbe