iip-ecosphere/platform #149: Getting logs of Python Service

hardikdava, 2023-07-20:
Hello, I am trying to build a python based AI service. I have following querries regarding MQTT based client and sanity check.

- Where is mqtt topic names are defined in [examples.python](https://github.com/iip-ecosphere/platform/tree/main/platform/examples/examples.python/`) ?
- Where/How can I get logs of python AI service? 

Thanks in advance.

eichelbe, 2023-07-20:
Hi,
currently, the logs can only be accessed via command line, but a UI-based solution is on the way (#147). Depending on your deployment (plan), installation and whether you use containers, please go to the device that is executing the service and open the service manager log (e.g., in a container, this is /var/log/iip.logs, running this part as systemd service the log would be in journald). In that log, you will find the trace of starting your application. Please navigate to the entries for that specific service. On line created by the spring service deployer indicates the specific location of the logging folder of that service. Currently, access to the logs is rather low-level, also due to the potential distributed execution of an application (@AhmadAlamoush @dzikaswinia please ensure that the most recent version of the platform handbook includes that information, #133). While for first development steps this is still ok (but complicated), it does not scale at all and is not comfortable. Thus, @dzikaswinia is working on a UI integration, where the log for a certain service can be selected and directly be streamed into the management UI.

And regarding the first question (@AlexanWeb)... the examples.python does not use any connector, so there are no MQTT topic names. For transport, that example uses AMQP through the platform internal broker which is reserved for internal communication, i.e., the AMQP channel names are implicit and dynamic. Similarly, examples.pythonSync is using MQTT for transport, but broker and channels are internal. 

I hope that helps a bit. Please do not hesitate to add subsequent questions.

hardikdava, 2023-07-20:
Hello @eichelbe, thank you for your response. I am working on FL-computer vision application. We got a server with the help of @alexander-weber. I think he prepared everything locally. I want to send an image using python client to inside Python-AI-service using mqtt/amqp. Assuming it will be sending on specific topic name in the format of `base64string`. Then is there any example scipt or documentation which explain specifically how to get data inside plateform using such communication protocol? 

It would be nice to have such end to end example. This will be beneficial for the future.

AlexanWeb, 2023-07-20:
Hello @hardikdava, we should first distinguish between MQTT for outside to inside the platform. For the connection of your MQTT broker to the platform could look like this: 
"""
    MqttV3Connector exampleMQTTCon = {
        id = " ExampleMQTTCon ",
        name = " Example MQTT Con ",
        description = "",
        ver = "0.1.0",
        host = <your Broker IP>
        port = <your Broker Port>,
        
        input = {{type=refBy(Dummy)}}, //The datatype you would send to your broker you can leave this out
        output = {{type=refBy(ImageType)}}, //The datatype you get from the broker
        inChannel = <The Topic your broker is receiving on>, // Example: “Topic+” where the + wold mean every sub topic  you can leave this out 
        outChannel = <The topic your Broker is sending on>,
        inInterface = {type=refBy(Dummy)}, //The data how it is send to your borker (ideally same as in and output) you can leave this out
        outInterface = {type=refBy(ImageType)}, //The data how it is received from your broker -> referring to objects 
        traceRcv = TraceKind::TRACE,
        traceSent = TraceKind::TRACE,
        machineFormatter = JsonFormatter{}, //The parser for the transported 
        machineParser = JsonParser{}
    };
"""
The Jason parser assumes you are sending the data contains a string representing valid json.
This would go into the .../test/easy/AllServicesPartFlower.ivml of the flower example.

IF you are sending data on inside the platfrom between services, refere to the example.python or example.flower where the service passes the defined object. I.e. passing on a object from asynchronous  python: """   self.ingest(result) """" where result is a object of the defined type. Or on a Synchronous serivce """ return result""". These Commands will pass that object on to the next service in the mesh.

I Hope this helps. The exmaple from HM23 could be made public only after the EMO. Feel free to ask follow up questions.

dzikaswinia, 2023-08-21:
Hi, 
log streams are now available in the management UI (runtime -> running services).

eichelbe, 2023-08-21:
... also the platform handbook has been updated and a preview for the actual version is now in github. Please feel free to let us know about missing information, parts to clarify, etc.

closed 2023-08-21 by eichelbe